OpenAutomationTechnologies/openPOWERLINK_V2 #108: #2455: Terasic DE2-115 MN fails to reach operational with process application configuration

SudharsanSudhan, 2016-04-07:
Configuration: 
MN Configuration: mn-dual-hostif-gpio
CN Configuration: X20 bus couplers
Configuration used: 30Cns/252TPDO/104RPDO with 2ms/5ms/10ms cycle time

Steps to reproduce:
1. Configure the process application scenario with the 5ms cycle time and build the application.
2. Clean the generated files using "make clean" from the command line
3. With 10ms cycle time, rebuild the application using "./create-this-app" from the command line 

Issue: 
1. "make clean" command doesn't clean the configurations

zelenkaj, 2016-04-08:
Is the issue solved if you run in step 3 `./create-this-app --rebuild` instead?

SudharsanSudhan, 2016-04-15:
Yes, using the command ./create-this-app --rebuild solves this issue.

zelenkaj, 2016-04-18:
Btw: Wh do you run make clean first and then create-this-app? If there is no change of the .sopcinfo file, you don't need to run create-this-app / except you change the board.

zelenkaj, 2016-04-20:
This is no bug, maybe lack of documentation?

closed 2016-04-20 by zelenkaj