OpenAutomationTechnologies/openPOWERLINK_V2 #126: NIOS2 build issue when NIOS2_EIC_PRESENT is defined.

RomainNaour, 2016-06-07:
The NIOS2 oplk CN drv doesn't build on NIOS2 system if one or more EIC (Extended Interrupt Controller) are present.

In commit 55959ef77e23c9752f9e6d013330b23665b265bb alt_irq_pending() is used directly by
OPENMAC_GETPENDINGIRQ() macro but alt_irq_pending() is not always defined.

See sys/alt_irq.h:
/*
 * alt_irq_pending() returns a bit list of the current pending interrupts.
 * This is used by alt_irq_handler() to determine which registered interrupt
 * handlers should be called.
 *
 * This routine is only available for the Nios II internal interrupt
 * controller.
 */
#ifndef NIOS2_EIC_PRESENT
static ALT_INLINE alt_u32 ALT_ALWAYS_INLINE alt_irq_pending (void)
{
  alt_u32 active;

  NIOS2_READ_IPENDING (active);

  return active;
}

I'm not sure if this is safe to do this in openmac-nios2.h:

#ifndef NIOS2_EIC_PRESENT
#define OPENMAC_GETPENDINGIRQ()                             alt_irq_pending()
#else
#define OPENMAC_GETPENDINGIRQ()                             0
#endif

Thoughts ?

zelenkaj, 2016-06-07:
Hi Romain,

is there no similar function for the EIC to get the pending irqs? Details about the reason for checking the pending irq before returning from the ISR you can find (THX to Michael) in 55959ef's commit message.

Does the EIC support edge triggered irqs? This could help you as well!

Best regards,
Joerg

RomainNaour, 2016-06-07:
Hi Joerg,

Thanks for the quick answer! And Thanks to Michael for the commit comment.

From what I can see there is no such function provided by the Altera API when the EIC is used.
Also, the 55959ef77e23c9752f9e6d013330b23665b265bb 's seems a timing issue with the IIC when used with NIOSII/e core.

We updated cn_pcp.qsys and cn_host.qys to use the NIOSII/f core plus an EIC + altera VIC.
In this context, I'm not sure that we are affected by the timer IRQ issue since the IRQ priorities are handled by the VIC.

Also, how trigger this issue ? how did you debug it ?

Best regards,
Romain

zelenkaj, 2016-06-07:
Hi Romain,

we noticed the issue in previous versions where the interrupt was acknowledged late in the ISR. Thus, the change 1b95725 moved the ack to the beginning of the ISR. However, to be sure that the ack has propagated the 55959ef introduced the barrier.

I don't know the EIC in details but I assume that the ISR is reentered if the corresponding IRQ is still asserted (if configured as level-triggered).

Best regards,
Joerg

RomainNaour, 2016-06-09:
Hi Joerg,

I don't know much nios2 systems than my colleague who customized the nios2 design but I think we can define OPENMAC_GETPENDINGIRQ() with IORD_ALTERA_VIC_INT_PENDING(PCP_0_VIC_0_BASE).

As I said previously, my system use an EIC + altera VIC, so the IORD_ALTERA_VIC_INT_PENDING() macro should do the same things than alt_irq_pending(). Also, I must provide the VIC base address.
So the OPENMAC_GETPENDINGIRQ implementation can be design dependent and it can be difficult to merge the change in master.

I need to do some test but at least the CN start correctly.

I'll send a RFC patch.

Best regards,
Romain

RomainNaour, 2016-06-09:
See https://github.com/RomainNaour/openPOWERLINK_V2/tree/develop-2.4-nios2-EIC

RomainNaour, 2016-06-13:
Hi Joerg,

Ok, I'll update the patch to avoid changing oplkcfg.h.

As stated in the nios2 document ug_embedded_ip.pdf, "The VIC can provide interrupt performance four to five times better than the nios2 processor's default internal interrupt controller (IIC)"
In our design, It's mainly to reduce the interrupt overhead.

I have to ask to my colleague how much is reduced the interrupt overhead.

Best regards,
Romain

zelenkaj, 2016-06-13:
Hello Romain,

the achievable reduction of the overhead would be interesting compared to the resource costs (LE).

Best regards,
joerg

mhogger, 2016-06-13:
Hello All,

I did an investigation on exactly this topic once, and it showed the saved time compared to the increase of LEs does not pay off.
But this was on a much older Quartus version - a up-to-date result would be interesting.

Best regards,
Michael

zelenkaj, 2016-06-14:
... was it with exception / interrupt stack linked to tightly-coupled memory and with using shadow registers? Romain, please let us know along with the results (area and performance) the used Quartus version.

RomainNaour, 2016-06-14:
Hi Joerg,

I'm using Quartus 14.0.

The cpu core is a NIOSII/f which is the only one with EIC an Avalon Streaming Sink input.
The Avalon Streaming Sink input is wired to the VIC interrupt_controller_out (see chapter 30-22 in ug_embedded_ip.pdf)

Also this nios2 core require a license to be used in a design.

There are some shadow register enabled in the Interrupt controller configuration and the Exception vector is linked with the tightly-coupled memory on the pcp processor.

Actually, we followed some design recommendations, and the target is to reduce the interrupt latency down to one or two dozen CPU cycles. I don't have the exact results right now.

Best regards,
Romain

RomainNaour, 2016-08-26:
Hi,

This issue has been fixed in develop-2.5 branch by da50e21426c6b717eedfd4cb2e9b78f23a01cbab.

Thanks!

Best regards,
Romain

closed 2016-08-26 by RomainNaour