OpenAutomationTechnologies/openPOWERLINK_V2 #128: Error initializing powerlink stack after application crashes

aghazi, 2016-06-09:
I am getting the following error when I try to call the function oplk_init() if my application does not close properly, and I try to run it again.

`"Resource could not be created (Windows, PxROS, ...)"`

It there anyway to reset the stack before starting the application in case of incorrect closure of application? I am using the 3.12.26-rt40 linux kernel and openPOWERLINK-V2.0.1. I also get the same error with V2.3.1.

banickn, 2016-06-20:
Hi aghazi,
I had a quick look and was able to reproduce the issue. I'll post here once i'm able to track it down further.

banickn, 2016-06-28:
Hello @aghazi

when you skip ` shutdownPowerlink()` it also doesn't invoke the `oplk_destroy()` function. When you have a look at the documentation of that function, it says the following:

> After cleaning up the openPOWERLINK stack with calling this function it is possible to re-create the openPOWERLINK stack with oplk_create.

Thus, what you can do is to just call the oplk_destroy function in your application before re-creating it.

banickn, 2016-07-20:
Hello @aghazi 

from our point of view the issue is resolved. Would you please have a look and confirm the solution I pointed out in my last post so we can close this issue up.

RomainNaour, 2016-09-15:
Hi all,

I think I have the same issue with my i210 Ethernet controller. 

If the cable is removed and the openpowerlink application is started, the stack will stop the first time with: oplk_create() with "Resource could not be created (Windows, PxROS, ...)" (0x0008)
Ok it's what I expected.

The cable is reconnected and the application is started again.
The stack is still failing with the same error:
oplk_create() with "Resource could not be created (Windows, PxROS, ...)" (0x0008)

The application is started now for the third time and "most of the time" the openPOWERLINK stack is started successfully.

Here is the dmesg:
First start
    [  600.693212] PLK: + powerlinkOpen...
    [  600.693222] PLK: + powerlinkOpen - OK
    [  600.699431] Registering Driver.....
    [  600.699443] bus: 'pci': add driver plk
    [  600.699467] bus: 'pci': driver_probe_device: matched device 0000:03:00.0 with driver plk
    [  600.699471] bus: 'pci': really_probe: probing driver plk with device 0000:03:00.0
    [  600.699599] edrvInstance_l.pIoAddr :0Xfc400000
    [  600.717807] Register MSI-X
    [  600.717847] plk 0000:03:00.0: irq 88 for MSI/MSI-X
    [  600.717857] plk 0000:03:00.0: irq 89 for MSI/MSI-X
    [  600.717866] plk 0000:03:00.0: irq 90 for MSI/MSI-X
    [  600.717897] ...Done
    [  600.718053] {initTxQueue}:Allocating Tx Desc f39c7800
    [  600.718058] ... Done
    [  600.718062] {initTxQueue}:Allocating Tx Desc f39c7000
    [  600.718066] ... Done
    [  600.718121] {initRxQueue}:Allocating Rx Desc f39c6800
    [  600.718125] ... Done
    [  600.718270] {initRxQueue}:Allocating Rx Desc f39c6000
    [  600.718275] ... Done
    [  600.718468] Configure Tx Queue 0Poll TXDCTL....Done
    [  600.728487] Configure Tx Queue 1Poll TXDCTL....Done
    [  600.738503] Configure Rx Queue 0Poll RXDCTL....Done
    [  600.748521] Configure Rx Queue 1Poll RXDCTL....Done
    [  600.758751] Requesting Interrupt...Done
    [  600.758945] initOnePciDev waiting for link up...Link Down
    [  603.807899] initOnePciDev finished with -5
    [  603.807909] plk: probe of 0000:03:00.0 failed with error -5
    [  603.807951] edrv_init pPciDev=NULL
    [  603.807955] edrv_exit calling pci_unregister_driver()
    [  603.807964] bus: 'pci': remove driver plk
    [  603.807978] driver: 'plk': driver_release
    [  603.809411] PLK: + powerlinkRelease...
    [  603.809419] PLK: + powerlinkRelease - OK

The second start:
    [  616.612376] PLK: + powerlinkOpen...
    [  616.612386] PLK: + powerlinkOpen - OK
    [  616.619877] PLK: + powerlinkRelease...
    [  616.619885] PLK: + powerlinkRelease - OK
I don't know what's going on, it seems that the i210 driver fail early with kCtrlInitStack command (initStack()) for some reason.

The third start:
    [  640.162715] PLK: + powerlinkOpen...
    [  640.162724] PLK: + powerlinkOpen - OK
    [  640.168972] Registering Driver.....
    [  640.168983] bus: 'pci': add driver plk
    [  640.169008] bus: 'pci': driver_probe_device: matched device 0000:03:00.0 with driver plk
    [  640.169012] bus: 'pci': really_probe: probing driver plk with device 0000:03:00.0
    [  640.169133] edrvInstance_l.pIoAddr :0Xfc500000
    [  640.187728] Register MSI-X
    [  640.187770] plk 0000:03:00.0: irq 88 for MSI/MSI-X
    [  640.187780] plk 0000:03:00.0: irq 89 for MSI/MSI-X
    [  640.187789] plk 0000:03:00.0: irq 90 for MSI/MSI-X
    [  640.187819] ...Done
    [  640.187973] {initTxQueue}:Allocating Tx Desc f39c6000
    [  640.187978] ... Done
    [  640.187982] {initTxQueue}:Allocating Tx Desc f39c6800
    [  640.187986] ... Done
    [  640.188041] {initRxQueue}:Allocating Rx Desc f39c7000
    [  640.188048] ... Done
    [  640.188192] {initRxQueue}:Allocating Rx Desc f39c7800
    [  640.188198] ... Done
    [  640.188387] Configure Tx Queue 0Poll TXDCTL....Done
    [  640.198406] Configure Tx Queue 1Poll TXDCTL....Done
    [  640.208423] Configure Rx Queue 0Poll RXDCTL....Done
    [  640.218440] Configure Rx Queue 1Poll RXDCTL....Done
    [  640.228671] Requesting Interrupt...Done
    [  640.228865] initOnePciDev waiting for link up...Link Up
Success!

I'm using the v2.4.1 stack on MN and CN.
Do you have an idea ? maybe the i210 controller is in unstable state after the link up timeout ?

Best regards,
Romain

RomainNaour, 2016-09-15:
Also adding opk_destroy() at the beginning ok the application, doesn't fix the issue.

RomainNaour, 2016-09-28:
Hello,

I finally had time to look closely to this issue.

So, after the first crash due to linkup error, the second crash is due to ioctl kCtrlInitStack error since the oplk_destroy() has not be called by the application.

With debug enabled in the oplk driver, we can see that this error is trigged by eventkcal_initQueueCircbuf() from eventkcal_init().
[  454.887646] Initialize kernel modules...
[  454.887651] eventkcal_initQueueCircbuf() Error: instance of queue 0 not NULL!
[  454.887654] eventkcal_init() Initialization error!

It appear that oplk_create() let the oplk Linux driver half initialized after the first crash.
The only way to delete all eventkcal buffer is indeed to call oplk_destroy().

So I fixed this error by calling unconditionally oplk_destroy() and oplk_exit() even if oplk_create() returned an error. (Previously only oplk_exit() was called).

So from my point of view the issue is resolved :)

Also, I would like to report another issue with the link up timout used in i210 openPOWERLINK driver.
It seems this timeout (EDRV_LINK_UP_TIMEOUT) is too small.
From time to time, the edrv i210 reach the link up timeout (3sec), thus the openpowerlink stack fail to initialize correctly and lead to the error reported above.
By measuring the link up time, it appear that the nominal link up time is very close to the timeout (2.8sec).
Increasing the timeout to 6sec in order to reach the timeout only if the Ethernet cable is not connected.

I'll send a pull request for this.

Best regards,
Romain

banickn, 2016-10-03:
Hello Romain,

I wasn't able to reproduce your issue. Decreasing EDRV_LINK_UP_TIMEOUT to a small value that it always fails didn't lead to the eventkcal_initQueueCircbuf() you mentioned in your last post but instead the stack was shut down properly every time. This was tested with the latest develop-2.5 branch and v2.3.2.

However, if anyone experiences similar initialization issues the way to go is to call oplk_destroy() and oplk_exit() as mentioned in the posts before.

Best regards,
Norman

zelenkaj, 2016-10-03:
Let's close this issue before we spend more time on this :-)

closed 2016-10-03 by zelenkaj