OpenAutomationTechnologies/openPOWERLINK_V2 #132: Dualprocshm compiler warnings

zelenkaj, 2016-06-16:
When compiling platforms with the dualprocshm CAL implementation several warnings are printed.

* Warnings for Altera C5 SoC ARM and Nios II:
[warn-dualprocshm_c5soc.txt](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/317903/warn-dualprocshm_c5soc.txt)
[warn-dualprocshm_nios2.txt](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/317905/warn-dualprocshm_nios2.txt)

* Warnings for Xilinx Microblaze and Zynq ARM:
[warn-dualprocshm_xilinx.txt](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/317904/warn-dualprocshm_xilinx.txt)

zelenkaj, 2016-06-16:
I've already fixed the warnings for Nios II, refer to a1ae7f1e70be4770628e1c096f56eb75118ee0d6 as reference for C5 SoC / Zynq ARM and Microblaze!

gauravks, 2016-06-17:
Thanks, I will prepare the fix for the ARM on C5SoC and Mibcroblaze and ARM on Zynq

zelenkaj, 2016-06-29:
#153

closed 2016-06-30 by zelenkaj