OpenAutomationTechnologies/openPOWERLINK_V2 #140: MN demo application compilation on 32 bit Windows with NDIS drivers

gauravks, 2016-06-21:
Kalycito bug Id: 2571
Bug Type:ENHANCEMENT
Design: Windows Terasic De2i-150 
Target Release: NA

The MN demo application (demo_mn_console) fails to compile on a 32 bit Windows system while using the Kernel stack on PCIe card or Windows Kernel Module stack libraries with following error

    > error LNK1120: 1 unresolved externals
    > error LNK2019: unresolved external symbol _CancelIoEx referenced in function _system_stopSyncThread
    > (apps\demo_mn_console\build\windows\system-windows.obj) demo_mn_console
    > warning C4013: CancelIoEx undefined; assuming extern returning int (apps\common\src\system\system-windows.c) demo_mn_console

The error is produced due to missing compiler options to specify the required 32 bit version of the system library.

zelenkaj, 2016-06-21:
Hello Gaurav,

please let me know the possible milestone for this issue!

Best regards,
Joerg

gauravks, 2016-06-21:
We can plan it for 2.5.0-rc

seissw, 2017-08-28:
Fix is provided and shall be included into next release.

closed 2017-08-28 by seissw