OpenAutomationTechnologies/openPOWERLINK_V2 #141: MN fails to reach Operational state with higher PDO configuration on Zynq emacPS

gauravks, 2016-06-21:
Kalycito bug Id: 2679
IssueType: BUG
Design: Linux Emacps (xilinx-z702)
Target Release: v2.5.0-rc1

**Configuration**:

MN | CN | CDC 
-------------------------- | ------------------- | -----------------------
demo-mn-console | Bus couplers | 30CN/252TPDO/104RPDO/10ms

The Linux openPOWERLINK MN using edrv-emacps on Zynq fails to run with above configuration resulting in following error

```
EVENT HISTORY Source = EventSourceEventk (0x06) OplkError = No free entry in internal buffer table for Tx frames (0x014) OrgSource = EventSourceNmtk 0x02
EVENT STATE_CHANGE   NmtMsNotActive->NmtGsOff Originating event:NmtEventCriticalError
```

powerlink-team-kalycito, 2017-01-18:
With the pull request #206, MN reaches operational state but when CNs are connected collision occurs and CNs resets. Hence the performance cannot be still measured for process application scenario (30CNs/252TPDO/104RPDO).

zelenkaj, 2017-01-18:
Collisions? Sounds like wrong pres timeout configuration...

powerlink-team-kalycito, 2017-02-03:
Collisions are removed with increased PRes timeout configuration. This bug can be closed.

closed 2017-02-03 by zelenkaj