OpenAutomationTechnologies/openPOWERLINK_V2 #143: PCP crash on APC2100 with higher PDO size

gauravks, 2016-06-21:
Kalycito bug Id: 2696
Bug Type: PERFORMANCE optimization
Design:Windows/Linux PCIe
Target Release: v2.4.1
Status: IN-PROGRESS

**Configuration:**

MN | CN | CDC
----------------------- | ---------------------------- | --------------------------------
demo_mn_console | Bus coupler | 30/252TPDO/104RPDO/2ms/5ms/10ms

When run with the above configuration, the PCP stops with no errors reported. The log for the failure case is as below:

```
Error Log:
2016/05/27-15:53:01 INFO     PDOMAP           0x1a11/20 Mapped object 0xA1C3/ 18
2016/05/27-15:53:01 INFO     PDOMAP           0x1a11/21 Mapped object 0xA1C3/ 19
2016/05/27-15:53:01 INFO     PDOMAP           0x1a11/22 Mapped object 0xA1C3/ 20
2016/05/27-15:53:01 INFO     PDOMAP           0x1a11/23 Mapped object 0xA1C3/ 21
2016/05/27-15:53:01 INFO     PDOMAP           0x1a11/24 Mapped object 0xA1C3/ 22
2016/05/27-15:53:01 INFO     PDOMAP           0x1a11/25 Mapped object 0xA1C3/ 23
2016/05/27-15:53:01 INFO     PDOMAP           0x1a11/26 Mapped object 0xA1C3/ 24
2016/05/27-15:53:01 EVENT    STATE_CHANGE   NmtGsResetCommunication->NmtGsResetConfiguration Originating event:NmtEventEnterResetConfig
Kernel stack has gone! Exiting...
2016/05/27-15:53:02 FATAL    CONTROL        Kernel stack has gone! Exiting...
Press Enter to quit!
Stack is in state off ... Shutdown
2016/05/27-15:53:40 INFO     CONTROL        Stack is in state off ... Shutdown openPOWERLINK
Exiting Sync Thread
```

zelenkaj, 2016-06-21:
Also reported at openPOWERLINK_APC2100: https://github.com/OpenAutomationTechnologies/openPOWERLINK_APC2100/issues/10

gauravks, 2016-07-15:
Observations from Debugging:

1. Heap and stack overlap is occurring in case the HEAP limit i.e. 256 KBytes is crossed.
2. Used stack checks in the BSP settings to verify the overlaps but there was no failure reported event during the error condition.
3. Checked by reducing the ALT_MAX_HEAP_BYTES to 2048 and allocating a memory of 1024 and 2048, in first case the memory was allocated but the second case malloc returned a NULL pointer as expected.
 * But with POWERLINK the PDO memory allocation exceeds the memory limit of 256 KBytes but no error was reported.
 * While using the debug mode in NIOS to EDS we noticed that after the execution of statement shown below the instruction pointers reset to 0x0 and processor hangs.
[image]
 The memory allocation in the above section is invoked for the PDO memory.

 * The hang state is shown in the following image
[image]