OpenAutomationTechnologies/openPOWERLINK_V2 #145: openCONFORMANCE  test fails with “GenericTimeoutException”

gauravks, 2016-06-21:
Kalycito bug Id: 2701
Issue Type: Bug
Design: All CN designs
Target Release: v2.4.1

Following error is reported by openCONFORMANCE while performing the test with CNs'
```
“GenericTimeoutException”
Timeout of 2000ms 0us 0ns expired.
```

mhogger, 2016-06-22:
Hello Gaurav,

Please provide some more information:
- When and how often does this test occur
- How can it be reproduced?
- Screenshot?

Thanks and regards
Michael

gauravks, 2016-06-22:
Hello Michael,

Please find the details below.

We performed the test 6 times and it was reproduced 5 times, the last time we were able to complete but with failures.

Test steps:
- Import the 00000000_POWERLINK_CiA401_CN.xdd from oplk\objdicts\CiA401_CN into the openCONFORMANCE
- Select the correct interface to be used for the test
- Start the test excluding advanced test

Test configuration.
MN: openCONFORMANCE at 80ms cycle time.
CN: avnet-s6plkeb (cn-single-gpio)

Screenshot:
[image]

The test report for the test is 
[00000000_POWERLINK_CiA401_TestResults_passed.zip](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/328072/00000000_POWERLINK_CiA401_TestResults_passed.zip)

Please let me know if you need any other details.

Regards,
Gaurav

zelenkaj, 2017-03-31:
This is an openCONFORMANCE issue, therefore closed.

closed 2017-03-31 by zelenkaj