OpenAutomationTechnologies/openPOWERLINK_V2 #148: Solve i210 prefetch issues (SoC jitter) on loaded systems

zelenkaj, 2016-06-22:
If the PC system is loaded heavily, the i210 cannot reliably hold the SoC jitter low.

zelenkaj, 2016-06-22:
Hello Gaurav,

please let me know the possible milestone when this could be solved!

Best regards,
Joerg

RomainNaour, 2016-09-29:
Hello Joerg,

I'm using the i210 Ethernet controller, can you some details about the jitter you have and your test case ?

Thanks!
Best regards,
Romain

zelenkaj, 2016-09-29:
Hello Romain,

usually the jitter with the i210 should be 240 ns. We noticed an increase of several µs (up to 6 µs) depending on the system and stress test (which should be not the case for i210 due to Qav support).

Best regards,
Joerg

RomainNaour, 2016-09-30:
Hello Joerg,

I measured the SoC jitter on my system with a Hilscher netAnalyzer without noticing any SoC jitter increase.
I'm using the same system that is currently used and tested on our hardware, I mean openpowerlink stack 2.3.2 on MN and CN. The i210 controler is used by the MN.
My powerlink cycle is 10ms and the measured SoC jitter is 210 ns max (tested during 2h50).
I can share with you the csv generated by the Hilscher tool, if you want to take a look.
Also the MN is leaded by running  stress -c 8 -i 4 -m 2 --vm-bytes 128M

Best regards,
Romain

banickn, 2017-03-22:
I ran the same tests (as Gaurav initially did) several times on multiple hardware setups and could not reproduce the SoC jitter mentioned in this issue. Lastly, I reduced the prefetch time of the i210 edrv and still didn't see any impact on the jitter.

closed 2017-03-22 by zelenkaj