OpenAutomationTechnologies/openPOWERLINK_V2 #149: Reduce timeout steps in ctrlucal_executeCmd for dualprocshm CAL

zelenkaj, 2016-06-22:
In order to improve the bootup time of platforms using dualprocshm the timeout steps shall be changed like it is done in this commit b6986511e68e7658a44781ee6acc59970372938a for hostif CAL.

zelenkaj, 2016-06-22:
Hello Gaurav,

I already did such change but the timeout of the C5 SoC ARM sleep function wasn't correct.

Best regards,
Joerg

gauravks, 2016-06-22:
Hello Joerg,

Do you have some information on the deviation which you had observed during the test?

Regards,
Gaurav

zelenkaj, 2016-06-23:
After doing the change for the dualprocshm the C5 SoC demo was not able to startup, because it ran into the short timeout. Sorry, but I don't know the exact deviation, haven't measured it.

gauravks, 2016-06-24:
okay, thanks for the info. We will check the implementation and make the necessary changes.

zelenkaj, 2017-01-23:
resolved by #211

closed 2017-01-23 by zelenkaj