OpenAutomationTechnologies/openPOWERLINK_V2 #152: hrtimer with preempt-rt

Math0, 2016-06-28:
Hello ! 

We have a bug on our installation with openPowerlink. We are using openPowerlink v2.3.2 and the Linux kernel 4.1.23 with preempt-rt 23. 
When a node is CN, we have a Kernel panic randomly:

```
[   94.407318] ------------[ cut here ]------------
[   94.407318] Kernel BUG at c00829ec [verbose debug info unavailable]
[   94.407348] Internal error: Oops - BUG: 0 [#1] PREEMPT ARM
[...]
[   94.407684] Backtrace: 
[   94.407714] [<c008283c>] (run_hrtimer_softirq) from [<c0045b74>] (do_current_softirqs+0x1ec/0x2a4)
[   94.407745]  r10:00000020 r9:c08086c0 r8:c07c2c38 r7:00000000 r6:04208140 r5:00000008
[   94.407745]  r4:c07c2d40 r3:c07c5178
[   94.407745] [<c0045988>] (do_current_softirqs) from [<c0046170>] (run_ksoftirqd+0x30/0x64)
[   94.407775]  r10:00000000 r9:00000002 r8:c07c2dec r7:00000001 r6:00000000 r5:dd814a00
[   94.407775]  r4:dd86a000
[   94.407806] [<c0046140>] (run_ksoftirqd) from [<c005f420>] (smpboot_thread_fn+0x160/0x290)
[   94.407836] [<c005f2c0>] (smpboot_thread_fn) from [<c005c11c>] (kthread+0xcc/0xe4)
[   94.407836]  r10:00000000 r9:00000000 r8:00000000 r7:c005f2c0 r6:dd814a00 r5:dd814a80
[   94.407867]  r4:00000000 r3:dd8592c0
[   94.407867] [<c005c050>] (kthread) from [<c0015d68>] (ret_from_fork+0x14/0x2c)
[   94.407897]  r7:00000000 r6:00000000 r5:c005c050 r4:dd814a80
[   94.407897] Code: e3130008 0affffb7 eb13d445 eaffffb5 (e7f001f2) 
[   94.715637] ---[ end trace 0000000000000002 ]---
[   94.715637] Kernel panic - not syncing: Fatal exception
```

After analyse, the bug is on kernel/time/hrtimer.c:hrtimer_rt_reprogram
This function tries to restart the timer but it is already active => Kernel Panic.
The callback hrtimer-linuxkernel.c:timerCallback provides The "restart" information.

Our correction is to check the state of the timer with hrtimer_active(pTimer_p) on  hrtimer-linuxkernel.c:timerCallback

```
Index: src/kernel/timer/hrestimer-linuxkernel.c
===================================================================
--- src/kernel/timer/hrestimer-linuxkernel.c	(révision 193705)
+++ src/kernel/timer/hrestimer-linuxkernel.c	(copie de travail)
@@ -473,6 +473,12 @@
             goto Exit;
         }
 
+        if(hrtimer_active(pTimer_p)) {
+        	printk("hrtimer_active !!!\n");
+            /* modified timer has already been restarted */
+            goto Exit;
+        }
+
 #ifdef PROVE_OVERRUN
         now      = ktime_get();
         interval = ktime_add_ns(ktime_set(0, 0), pTimerInfo->period);

```

Do you think our correction is correct ?

We have found another small problem:

```
Index: src/kernel/timer/hrestimer-linuxkernel.c
===================================================================
--- src/kernel/timer/hrestimer-linuxkernel.c	(révision 189443)
+++ src/kernel/timer/hrestimer-linuxkernel.c	(révision 189444)
@@ -212,7 +212,7 @@
 
     for (index = 0; index < TIMER_COUNT; index++)
     {
-        pTimerInfo = &hresTimerInstance_l.aTimerInfo[0];
+        pTimerInfo = &hresTimerInstance_l.aTimerInfo[index];
         pTimerInfo->pfnCallback = NULL;
         pTimerInfo->eventArg.timerHdl = 0;
         /* In this case we can not just try to cancel the timer.

```

banickn, 2016-06-29:
Hi @Math0
I will have a look and try to reproduce your problem. As a side note, we are currently only running openPOWERLINK tests on linux kernel 3.12, so unfortunately compatibility with more recent kernel versions can't be guaranteed right now.

However, you meant kernel 4.1.20 with rt-23, didn't you? Just wondering because I can't find a rt-patch for 4.1.23.

Math0, 2016-06-29:
We were running the stack on linux kernel 3.2 but we had to upgrade to 4.1 for RT issue.
We are using RT TI kernel: https://git.ti.com/ti-linux-kernel/ti-linux-kernel branch: ti-rt-linux-4.1.y
Apparently it's a 4.1.20 kernel with his rt-patch, upgraded afterwards in 4.1.23

Our board is a BeagleBone Black variation. We developped our own edrv Kernel driver, inspired from the original Ethernet driver (http://lxr.free-electrons.com/source/drivers/net/ethernet/ti/cpsw.c?v=4.1) and your others edrv driver. we can share it if needed.

banickn, 2016-06-30:
Hi @Math0 
Alright, that explains it. I ran some tests with a 4.1.20-rt23 kernel and our edrvs and couldn't spot the problems you're facing with your setup.

Nonetheless, would you please submit a "Pull request" on github with your hrestimer fix so we can merge it into openPOWERLINK. We would definitely be interested in your BeagleBone edrv!
Please submit that as a separate pull request. After that is done we will need some time to review it to make sure it integrates smoothly into the stack.

sinseman44, 2016-07-05:
Hello,
I work with @Math0 on Powerlink Stack and would suggest a patch to solve it.

To complete what @Math0 said, the problem comes on a RMN/CN device, when the stack restarts the cycle timer on SoC frame reception (file _dllkframe.c_, line _2167_). This timer is configured, automatically, with the parameter :

> fContinue_p

 equal to true in function edrvcyclic_startCycle.

    #if defined(CONFIG_INCLUDE_NMT_RMN)
        if (dllkInstance_g.fRedundancy)
        {
            if (nmtState_p == kNmtCsOperational)
            {
                 hrestimer_modifyTimer(&dllkInstance_g.timerHdlSwitchOver,
                                                      dllkInstance_g.dllConfigParam.switchOverTimeMn * 1000ULL,
                                                      dllk_cbTimerSwitchOver, 0L, FALSE);
             }
             else
             {
                  hrestimer_modifyTimer(&dllkInstance_g.timerHdlSwitchOver,
                                 dllkInstance_g.dllConfigParam.delayedSwitchOverTimeMn * 1000ULL,
                                 dllk_cbTimerSwitchOver, 0L, FALSE);
             }
             if ((ret = edrvcyclic_startCycle()) != kErrorOk)
                return ret;
        }
        else
    #endif

on linux kernel context, in hrtimer timeout callback (file _hrestimer-linuxkernel.c_, function _timerCallback_), sometimes when the stack is preempted, the cycle hrtimer is restarted (hrtimer state is enqueued) after the test of timer handle.
the goto instruction is not called because the two handles are identical.

    if (orgTimerHdl != pTimerInfo->eventArg.timerHdl.handle)
    {
        /* modified timer has already been restarted */
        goto Exit;
    }

This callback returns HRTIMER_RESTART while hrtimer state is not HRTIMER_STATE_INACTIVE and the kernel panic arrives in _hrtimer_rt_reprogram_ function.

In case of the device is configured as RMN/CN, I think it's not necessary to start a cyclic hrtimer each time the powerlink stack received a SoC frame.

    Index: include/kernel/edrvcyclic.h
    =============================================
    --- include/kernel/edrvcyclic.h
    +++ include/kernel/edrvcyclic.h
    @@ -106,7 +106,7 @@
     tOplkError edrvcyclic_init(void);
     tOplkError edrvcyclic_exit(void);
     tOplkError edrvcyclic_setCycleTime(UINT32 cycleTimeUs_p, UINT32 minSyncTime_p);
    -tOplkError edrvcyclic_startCycle(void);
    +tOplkError edrvcyclic_startCycle(BOOL fredudancy_p);
     tOplkError edrvcyclic_stopCycle(BOOL fKeepCycle_p);
     tOplkError edrvcyclic_setMaxTxBufferListSize(UINT maxListSize_p);
     tOplkError edrvcyclic_setNextTxBufferList(tEdrvTxBuffer** ppTxBuffer_p, UINT txBufferCount_p) SECTION_EDRVCYC_SET_NEXT_TX;

    Index: src/kernel/edrv/edrvcyclic.c
    =============================================
    --- src/kernel/edrv/edrvcyclic.c
    +++ src/kernel/edrv/edrvcyclic.c
    @@ -307,12 +307,15 @@
 
     This function starts the cycles.
 
    +\param  fredudancy_p    if TRUE, start cyclic timer,
    +                        otherwise start one shot timer
    +
     \return The function returns a tOplkError error code.
 
[... 42 lines cut ...]
                             return ret;
 
                         // initialize cycle counter

If the powerlink stack detects a lost of MN, SwitchOver hrtimer callback send AMNI frame, post an event to be the new AMN (new Nmt State : kNmtMsOperational) and restarts the cycle hrtimer in continuous mode  : 

    Index: src/kernel/dll/dllkevent.c
    ==========================================
    --- src/kernel/dll/dllkevent.c	(révision 194495)
    +++ src/kernel/dll/dllkevent.c	(copie de travail)
    @@ -500,6 +500,12 @@
                     if ((ret = hrestimer_deleteTimer(&dllkInstance_g.timerHdlSwitchOver)) != kErrorOk)
                         return ret;
 
    +                /* start cycle timer to send frames */
    +                ret = edrvcyclic_startCycle(TRUE);
    +                if(ret != kErrorOk) {
    +                    return ret;
    +                }
    +
                     dllkInstance_g.socTime.relTime += dllkInstance_g.dllConfigParam.cycleLen;
                     // initialize SoAReq number for ProcessSync (cycle preparation)
                     dllkInstance_g.syncLastSoaReq = dllkInstance_g.curLastSoaReq;

I test this with two devices (1 AMN and other RMN) and it works well.

banickn, 2016-07-20:
Hello @Math0 and @sinseman44 
would you please use the "pull request" feature here on github to submit your patches?
This would allow us to review your patches properly and integrate it into the stack if possible.

Here is a very decent manual how to use it: https://help.github.com/articles/using-pull-requests/

zelenkaj, 2016-08-16:
Hello everyone,

What is the state of this issue?

@Math0 and @sinseman44 can you please share your patches preferably as pull requests? You could also provide .patch files attached to this issue if you prefer. Btw, what is the status of your activity with the BBB?

Thanks,
Joerg

sinseman44, 2016-08-31:
Hello everyone,
The state of this issue will be closed, I will share my patches as soon as possible.

Sinseman44

banickn, 2016-12-05:
Hi sinseman44,
we would still be quite pleased if you could submit your patches, so this issue can be closed officially.

Thanks and best regards,
Norman

sinseman44, 2016-12-05:
Hi Norman,
Ok, i'll submit my patches this week. sorry for the delay.

Sinseman44

closed 2017-02-02 by zelenkaj