OpenAutomationTechnologies/openPOWERLINK_V2 #154: #2753: Warning issue on Linux 8255x edrv

SudharsanSudhan, 2016-07-06:
Stack version     : 2.4.1-rc1
SHA                    : 4ab489e4a617474f61446e784ed6c84a3d591889
Test case            : Compile test
Managing Node  : Linux edrv – 8255x

When compiling the driver for Linux edrv (8255x), some warnings are raised:

Following warning message has appeared in the console log:
_<openPOWERLINK_dir>/drivers/linux/drv_kernelmod_edrv/build/src/edrv-8255x.c: In function ‘edrvIrqHandler’:
<openPOWERLINK_dir>/drivers/linux/drv_kernelmod_edrv/build/src/edrv-8255x.c:924:38: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
                                      (dma_addr_t)pDmaBuffer,
                                      ^
<openPOWERLINK_dir>/drivers/linux/drv_kernelmod_edrv/build/src/edrv-8255x.c: At top level:
cc1: warning: unrecognized command line option "-Wno-date-time" [enabled by default]_

zelenkaj, 2016-07-06:
I assume that you compile on 64 bit Linux, right? What kernel version do you use?

SudharsanSudhan, 2016-07-06:
No, we are using 32 bit Linux for compilation and kernel version is 3.12.24-rt38.

zelenkaj, 2016-07-06:
Okay, now I know why I couldn't see the warning... Thx!

zelenkaj, 2016-07-06:
Gaurav, can you please let me know the possible milestone.

gauravks, 2016-07-06:
I checked and the following issue
```
/drivers/linux/drv_kernelmod_edrv/build/src/edrv-8255x.c: At top level:
cc1: warning: unrecognized command line option "-Wno-date-time" [enabled by default]
```

seems to be introduced by https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/pull/61. As this commits is valid for Linux kernel > 3.14, I will check alternatives.

The first warning can be fixed for 2.4.1
```
/drivers/linux/drv_kernelmod_edrv/build/src/edrv-8255x.c: In function ‘edrvIrqHandler’:
/drivers/linux/drv_kernelmod_edrv/build/src/edrv-8255x.c:924:38: warning: cast from pointer to integer of different size -Wpointer-to-int-castpDmaBuffer,
```

zelenkaj, 2016-07-06:
Okay, can you please move the "-Wno-date-time" warning to a separated issue.

gauravks, 2016-07-11:
The fix for the first part is share in #157. Apparently the "-Wno-date-time" warnings are also not generated after these changes.

zelenkaj, 2016-07-11:
Can I close the issue or is there still some pending activity ?

gauravks, 2016-07-12:
Yes, the issue can be closed.

closed 2016-07-12 by zelenkaj