OpenAutomationTechnologies/openPOWERLINK_V2 #156: #2756: Warning issue on Zynq on Linux

SudharsanSudhan, 2016-07-06:
Stack version  : 2.4.1-rc1
SHA            : 4ab489e4a617474f61446e784ed6c84a3d591889
Test case      : Compile test
Managing Node  : Zynq on Linux

When compiling the stack for Zynq on Linux, some warnings are raised:

Following warning message is appeared in the console log:
_<openPOWERLINK-Dir>/stack/src/user/event/eventu.c: In function 'eventu_process':
<openPOWERLINK-Dir>/stack/src/user/event/eventu.c:347:5: warning: 'eventSource' may be used uninitialized in this function [-Wmaybe-uninitialized]
<openPOWERLINK-Dir>/stack/src/user/event/eventu.c:190:29: note: 'eventSource' was declared here_

zelenkaj, 2016-07-06:
good compiler :-)

Could be solved by assigning `kEventSourceInvalid` to `eventSource` in `eventu_process()` initially (or in default case).

gauravks, 2016-07-12:
Resolved by https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/pull/158

closed 2016-07-12 by zelenkaj