OpenAutomationTechnologies/openPOWERLINK_V2 #164: SDO segmented data transfer issue

RitishaRamesh, 2016-08-12:
### Test configuration:
Managing Node: Linux 8255x edrv
Controlled Node: Linux 8111 edrv
CDC configuration: Demo_3CN

### Following steps were carried out to perform the test:

- Change the MN application (<openPOWERLINK-dir>/stack/src/user/cfmu.c) to transfer data(Using .srec file) from MN to CN
- Change the following files in CN application (<openPOWERLINK-dir>/apps/demo_cn_console/src) to receive the data from MN  

        1. <openPOWERLINK-dir>/apps/demo_cn_console/src/event.c 
        2. <openPOWERLINK-dir>/apps/demo_cn_console/src/main.c
        3. Add fwupdate.c and fwupdate.h files in <openPOWERLINK-dir>/apps/demo_cn_console/src folder

- Run both MN and CN application
- Compare SDO data in both the .srec file of MN and the received .txt file of CN.    

### Observations:
- Linux Edrv design: (tested with 5ms cycle time)

        1. Mismatch of SDO data transfer from MN to CN        
        2. Instead of SDO data, 0's are transferred (Verified using network trace)
        3. Loss of SDO data (Verified from received file and network trace)
- Linux PCAP design:

        1. Segmented data transfer is not supported with 50ms cycle time  

### Following test files are attached for references(circbuff-files.zip)  
[circbuff-files.zip](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/415332/circbuff-files.zip)

-  Source files for Segmented data transfer test(Both MN and CN)
-  Transferred data file and received data file
-  Network trace

banickn, 2016-10-10:
Hello RitishaRamesh,

could you please provide your changes as patch file instead of separate *.c files?
That would make it easier to apply your changes and see the differences instantly.

Best regards,
Norman

RitishaRamesh, 2016-10-12:
Hello Norman,

Please find attached, the patch files for the changes made for both Managing Node and Controlled Node for performing SDO segmented data transfer test.

[sdo-segmented-transfer.zip](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/524562/sdo-segmented-transfer.zip)

Thanks & Regards,
Ritisha Ramesh

banickn, 2017-03-10:
Hi Ritishma,

could you please use any other edrv (e.g. i210 or 82573) instead of 8111?
Unfortunately we don't have the 8111 interface available over here so it's not possible for us to reproduce it.
Furthermore, it would be good if you could re-run the tests with the latest oplk v2.5.1 stack version.

Thanks!
Best regards,
Norman

powerlink-team-kalycito, 2017-08-22:
This issue is not reproducible in latest openPOWERLINK stack V2.6.0 (as firmware upgrade feature is integrated). In addition, we have successfully validated segmented data transfer via SDO frames in firmware upgrade test.

Hence this issue can be closed.

closed 2017-08-22 by seissw