OpenAutomationTechnologies/openPOWERLINK_V2 #170: Unable to configure openpowerlink  on externel PCIe

Chris-texas, 2016-10-21:
Unable to run the demo application when the kernel part of the openPOWERLINK stack is located on an external PCIe device. I am getting the following error. Could somebody help me with this?

Thanks 

Error:

system_init() couldn't set nice value! (Operation not permitted)
system_init() couldn't set thread scheduling parameters! 20
2016/10/21-13:00:06 INFO     GENERIC        demo_mn_console: Stack Version:V2.4.1 Stack Configuration:0x0000002B
2016/10/21-13:00:06 INFO     GENERIC        Using CDC file: mnobd.cdc
2016/10/21-13:00:06 INFO     CONTROL        Initializing openPOWERLINK stack
ctrlucal_init() open return error -1 (Unknown error -1)
Could not initialize ctrlucal
oplk_initialize() failed with "Resource could not be created (Windows, PxROS, ...)" (0x0008)
2016/10/21-13:00:06 FATAL    CONTROL        oplk_init() failed with "Resource could not be created (Windows, PxROS, ...)" (0x0008)

oplk_freeProcessImage() failed with "API called but stack is not initialized/running" (0x014e)
oplk_execNmtCommand() failed with "API called but stack is not initialized/running" (0x014e)
2016/10/21-13:00:06 INFO     CONTROL        Stack is in state off ... Shutdown openPOWERLINK
eventu_postEvent() Eventu module is not initialized
eventu_postEvent() Eventu module is not initialized
eventu_postEvent() Eventu module is not initialized
eventu_postEvent() Eventu module is not initialized
eventu_postEvent() Eventu module is not initialized
ctrlucal_executeCmd() ioctl error -1
Stack is in state off ... Shutdown`

Chris-texas, 2016-10-21:
asdas

Chris-texas, 2016-10-21:
Additional Info:
I am using Powerlink v 2.4.1 
Kernel version 3.10.0-327.28.3.el7.x86_64
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)

zelenkaj, 2016-10-21:
Do you use the Linux Edrv Kernel driver (refer to doc: http://openpowerlink.sourceforge.net/doc/2.4/2.4.1/page_build_drivers.html#sect_build_drivers_build_linux_edrv) or do you use pcap (with user space daemon or directly linked)?

zelenkaj, 2016-10-21:
Just to be sure (since you explicitly mentioned *external PCIe* in the issue header): **Don't** use the *Linux Kernel PCIe Interface Driver* which is intended for PCIe interface cards with FPGA implementing the openPOWERLINK kernel stack.

Chris-texas, 2016-10-22:
I am trying to set up communication between two linux PCs, one as MN and the other as CN.. 
I am using : "Building a Linux Kernel PCIe Interface Driver", I am not using FPGA board.
I had  also tried using the Linux Erdv Kernel driver, in a separate setup but when I run the demo application (console/qt) the pc reboots. 
I dont know what I am doing wrong.

zelenkaj, 2016-10-24:
With having no kernel stack running on an FPGA the *Building a Linux Kernel PCIe Interface Driver* makes no sense! This driver only works for B&R APC2100 with openPOWERLINK interface card or with TERASIC DE2i-150 evaluation board. It is not probing for the Ethernet controller you have!

There is an Ethernet driver implementation available for the Realtek 8111 (stack/src/kernel/edrv/edrv-8111.c) - this Ethernet driver is only used with the *Linux Edrv Kernel Driver*. When building the *Linux Edrv Kernel Driver* you can select the Realtek 8111!

For details please refer to the building documentation: http://openpowerlink.sourceforge.net/doc/2.4/2.4.1/page_build.html

Chris-texas, 2016-10-24:
As of now my task is to implement Powerlink Stack in three configurations  ( as a complete library and split libraries) and compare the performances. 

I was able to configure it as a complete library and in User Mode (where the openPowerlink kernel layer is running as a separate Linux user space daemon).

But I am unable to configure openPowerlink stack using ethernet driver (edrv) OR PCIe Interface driver.
-- I have followed the document and compiled/built Stack accordingly as per the procedure.
-- The communication is being set up between two PCs Centos 7. There is no FPGA board in picture. 
-- with edrv I do not get any error message when I run the demo application. However, the computer restarts as soon as I run the demo application.

-- when using PCIe Interface driver, I get an error message ( the message is posted above).
      -- Do I use plkload script while loading PCIe interface driver? I have tried both ways(with & without) ending up in the same result.
      -- I also tried debugging, I found that _open system call_ called in ctrlucal_init() is returning error because the file path given as a parameter does not exist.

[image]

#define PLK_DEV_FILE      "/dev/plk"

I checked the /dev directory there is no plk there.

I believe I am doing some silly mistake out of my ignorance. I really appreciate your time and patience with this issue.

Chris-texas, 2016-10-24:
kernel log after the module is loaded and powerlink demo is run..

[image]

zelenkaj, 2016-10-24:
Again, you cannot use the PCIe interface driver without having the supported hardware (FPGA interface card).

You need to use the *Linux Edrv Kernel Driver*:

- Build stack library with kernel interface (CFG_COMPILE_LIB_MNAPP_KERNELINTF = ON)
- Build Linux kernel module in `drivers/linux/drv_kernelmod_edrv` with edrv-8111.c (CFG_POWERLINK_EDRV = 8111)
- Build console demo with Linux Kernel Module option (CFG_BUILD_KERNEL_STACK = "Linux Kernel Module")

zelenkaj, 2016-10-24:
I close this issue, because the PCIe driver interface cannot work without having the needed hardware available.

Please create a new issue for the crashing demo application with using *Linux Edrv Kernel Driver*.

closed 2016-10-24 by zelenkaj

Chris-texas, 2016-10-24:
Thank you for the info.