OpenAutomationTechnologies/openPOWERLINK_V2 #174: PC reboots while running Application Demo using  Linux Edrv Kernel Driver.

Chris-texas, 2016-10-24:
I have followed the instructions to configure openPowerlink stack using  Linux Edrv Kernel Driver. However when I run the demo application the computer reboots. There is no error message.

I am using Powerlink v 2.4.1
Kernel version 3.10.0-327.28.3.el7.x86_64
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 07)

--Trying to use 03:00.0 Ethernet controller.

zelenkaj, 2016-10-24:
Have you checked any logs in /var/log ?

Chris-texas, 2016-10-24:
When I access kernel log after reboot, I find nothing. I am using dmesg -wH to view the logs.
There is a file name vmcore-dmesg.txt in /var/crash, but It looks very complicated. 
I have attached the file, if it helps.
[Doc.txt](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/547952/Doc.txt)

zelenkaj, 2016-10-24:
There is some useful information in the log! The crash happens because the edrv is not removing the pcidev correctly. However, openPOWERLINK reports:

~~~~
[ 1797.396293] initOnePciDev Unsupported chip! TCR = 0x2F900F80
~~~~

... the version of the NIC is not supported by the Ethernet driver - refer to stack/src/kernel/edrv-8111.c:1713
~~~~C
    // Check hardware version, i.e. chip ID
    temp = EDRV_REGDW_READ(RW_REGDW_TCR);
    if (((temp & REGDW_TCR_VER_MASK) != REGDW_TCR_VER_8111B_8168B) &&
        ((temp & REGDW_TCR_VER_MASK) != REGDW_TCR_VER_8100E) &&
        ((temp & REGDW_TCR_VER_MASK) != REGDW_TCR_VER_8101E) &&
        ((temp & REGDW_TCR_VER_MASK) != REGDW_TCR_VER_8111D))
    {
        // unsupported chip
        printk("%s Unsupported chip! TCR = 0x%08lX\n", __FUNCTION__, (ULONG)temp);
        result = -ENODEV;
        goto ExitFail;
    }
~~~~

You have these options:
- Use a different NIC that is supported by edrv-8111 or any other edrv (e.g. i210)
- Port the edrv-8111 to the NIC revision you have

Chris-texas, 2016-10-26:
Thanks for finding the root cause.

I will have to go with the second option. Few hints would be great as to how I begin with this. Are there any specific changes I have to make in edrv-8111.c?

zelenkaj, 2016-10-27:
First, I would check the differences of the hardware supported by edrv-8111 and the NIC you have. In the code I would start with temporarily bypassing the hardware version check that throws the unsupported chip error. If that all works fine, the final change will be by extending the hardware revision check by the NIC you have. Keep us posted!

Chris-texas, 2016-10-28:
Hello,
I commented out the result and goto statements in edrv-8111.c. The system does not reboot now.
[image]

I could find the register description for RTL8111B RTL8168B but I could not find register information for RTL8168.
I get the following msg via dmesg -
[image]

zelenkaj, 2016-11-07:
@powerlink-team-kalycito: This issue is about the PC reboot due to faulty exit on error handling. It is **not** about the the unsupported NIC revision!

Chris-texas, 2016-11-07:
Hello,

I have one more question. When I bypass the hardware version check, it gives me the below mentioned error: 
[image]

and the kernel log says: 
[image]

Could you please help me with that?

Thanks!

banickn, 2016-11-10:
Hi Chris,
to me it looks like that it fails during the mdio register writing part of the initOnePciDev function.
I would reckon that the PHY control register of your hardware version differ to the ones supported by the edrv-8111. Maybe you can find some implementation of your version and compare it as zelenkaj suggested few posts earlier.

zelenkaj, 2017-02-09:
Rescheduled after V2.5.1

powerlink-team-kalycito, 2017-03-29:
Hi Chris,

We don't have the right HW here. Could you comment if this issue is still pending?

Regards,
Powerlink-Team-Kalycito

seissw, 2017-07-20:
Since we are not able to reproduce the issue on our side and we did not get any feedback since several months, we consider this issue as being closed.

closed 2017-07-20 by seissw