OpenAutomationTechnologies/openPOWERLINK_V2 #179: MAC address for Zynq hybrid MN is displayed as 0

SudharsanSudhan, 2016-11-11:
Bug description:

Stack version  : 2.5.0-rc2
SHA            : 03d3974dd069673e5f04870e450e3f15e684ef9e
Test case      : API test - oplk_getEthMacAddr ( ) 
Managing Node  : **Zynq hybrid MN(mn-dual-shmem-gpio)**
Controlled Node: Bus controllers
Cycle time     : Demo_3Cn (50ms/1TPDO/1RPDO)

Bug summary:
MAC address for Zynq hybrid MN is displayed as (0: 0: 0: 0: 0: 0).

Steps to reproduce:
1. Build the stack, driver and application with oplk_getEthMacAddr ( )
2. Run the application

Console log:
--------------------------------------------------
openPOWERLINK console MN DEMO application
Using openPOWERLINK stack: V2.5.0-rc2
---------------------------------------------------
2016/06/09-16:34:56 INFO    GENERIC    demo_mn_console: Stack version:V2.5.0-rc2 Stack configuration:0x0000000B
2016/06/09-16:34:56 INFO    GENERIC    Using CDC file: mnobd.cdc
Initializing openPOWERLINK stack...
2016/06/09-16:34:56 INFO    CONTROL    Initializing openPOWERLINK stack

_**Ethernet Interface MAC address : 0: 0: 0: 0: 0: 0**_
Initializing process image...

zelenkaj, 2016-11-11:
The openMAC Ethernet driver assigns no MAC address instead it uses the MAC address provided when initializing the stack! Is there any issue if the MAC address is all zeros, except that the getter function returns all zeros? I mean, for this test case you could simply assign some MAC address others than all zeros and check it against the getter result.

RitishaRamesh, 2016-11-15:
Yes, we have checked that the MAC address getter functions reads the MAC address set by the application.

But VEth implementation for the Zynq hybrid design doesn't work if the MAC address is assigned all zeros.

zelenkaj, 2016-11-16:
"But VEth implementation for the Zynq hybrid design doesn't work if the MAC address is assigned all zeros." --> Create an extra issue for that.

zelenkaj, 2016-11-16:
This issue can be closed, the test case "API test - oplk_getEthMacAddr ( )" must be changed for targets that have no MAC address assigned in non-volatile memory (e.g. targets with openMAC).

closed 2016-11-16 by zelenkaj