OpenAutomationTechnologies/openPOWERLINK_V2 #180: Zynq emacPS MN fails to reach Operational state with higher PDO configuration

SudharsanSudhan, 2016-11-11:
Bug description:

Stack version  : 2.5.0-rc2
SHA            : 03d3974dd069673e5f04870e450e3f15e684ef9e
Test case      : Performance test
Managing Node  : xilinx-z702 (Emacps-edrv)
Controlled Node: Bus controllers

Configuration     : 
    1. Motion application test with 36TPDO/24RPDO/250us/400us/1000us
    2. IO application test with 64TPDO/64RPDO/250us/400us/1000us

Bug summary:
The Linux openPOWERLINK MN using edrv-emacps on Zynq fails to run with above configuration, resulting in following error.

Console log:
_EVENT   NODE        Node= 15, NmtNodeEventCheckChrestimer callb!onf State:NmtCsPreOperational1
hrestimer callback: Continuous timer (handle 0x10000001) had to skip 5 interval!
hrestimer callback: Continuous timer (handle 0x10000001) had to skip 4 interval!
hrestimer callback: Continuous timer (handle 0x10000001) had to skip 3 interval!_

zelenkaj, 2016-11-11:
Same issue: #141 ?

RitishaRamesh, 2016-11-15:
No, the issue is not the same. The configuration for this issue is Motion application (36TPDO/24RPDO/250us/400us/1000us) and IO application scenario (64TPDO/64RPDO/250us/400us/1000us) and we receive "hrestimer callback" error. And the configuration for #141 is Process application scenario (252TPDO/104RPDO/10ms) and we receive "No free entry in internal buffer table for Tx frames" error

zelenkaj, 2016-11-16:
Okay, thx for clarification. Maybe next time make it clear already in the issue's header (check if you can edit the header?).

powerlink-team-kalycito, 2017-01-18:
With the pull request #206, MN reaches operational state and 7 CNs are stable at 1ms in the given pull request for motion application scenario (10CNs/36TPDO/24RPDO) (Expected 10 CNs to be stable at 1ms). But MN resets on connecting CNs in the IO application scenario (20CNs/64TPDO/64RPDO). 

Hence the bug has to be re-opened.

powerlink-team-kalycito, 2017-03-23:
When retesting this issue in 2.5.1, The performance achieved for Zynq emacPS design is seen to be:

1) Motion application scenario (10CNs/36TPDO/24RPDO)  - working at 1ms
2) IO application scenario (20CNs/64TPDO/64RPDO) - working at 2ms

After running for few hours (>2 hrs) in this configuration,
"SDO Abort=SDO_AC_DATA_NOT_TRANSF_DUE_LOCAL_CONTROL(0x08000021), 
 Error=0x0000" is observed, causing CNs to be reset. Similarly, the error "HistoryEntry: Type=0x3002 Code=0x8243" is repeatedly observed.

Hence from performance point of view, this bug can be closed because the performance is much improved compared to previous versions.

zelenkaj, 2017-03-29:
Please create follow up issue, this issue is closed now.

closed 2017-03-29 by zelenkaj