OpenAutomationTechnologies/openPOWERLINK_V2 #181: Number prints are displayed in the console log

SudharsanSudhan, 2016-11-11:
Bug description:

Stack version  : 2.5.0-rc2
SHA        : 03d3974dd069673e5f04870e450e3f15e684ef9e
Test case      : Basic run test
Managing Node  : Hybrid Zynq MN (Linux on ARM + Microblaze)
Controlled Node: Bus controllers
Configuration  : Demo_3CN with 50ms/1TPDO/1RPDO

Bug summary:
Once the PDO Configuration is downloaded, number "2222222" is displayed in console log.

Steps to reproduce:
1. Build the stack, driver and demo application to perform the basic run test
2. Run the demo_mn_console application

Console log:
_2016/06/09-16:37:14 INFO PDOMAP 0x1601/16 Mapped object 0xA4C0/ 52
2016/06/09-16:37:14 INFO PDOMAP 0x2222222222222222222222222222222222222222222222_

Screenshot:
[image]

zelenkaj, 2016-11-11:
Check if this is a known issue for Xilinx Zynq platforms

powerlink-team-kalycito, 2017-04-04:
This issue is happening when more objects are mapped (mapped objects > 16 bytes). The root cause  of this issue is CPU idle time availability because the processor is loaded with fast and higher number of networking interrupts, compared to the slower UART interrupt for the serial port. So the console does not get a chance to display the text.

Since the issue is related to CPU performance and not network performance, this bug may be closed.

closed 2017-04-04 by zelenkaj