OpenAutomationTechnologies/openPOWERLINK_V2 #183: PRes Chaining fails to work with Demo_3Cn configuration

SudharsanSudhan, 2016-11-11:
Bug description:

Stack version  : 2.5.0
SHA            : ed45710542594400051c679fcf05f09e44b672bb
Test case      : PResChaining test
Platform       : All platform 
Controlled Node: Bus controllers
Configuration  : Demo_3CN with 10ms/Chained station(NodeID-1,32,110)/1TPDO/1RPDO.

Bug summary:
Controlled Nodes continuously resets without reaching operational state.

Steps to reproduce:
1. Build the stack, driver and demo application to perform PRes Chaining test
2. Run the MN application with the above mentioned configuration.

zelenkaj, 2016-11-11:
Please provide the configuration (openCONF project, cdc) used by the MN.

powerlink-team-kalycito, 2017-02-03:
Updated the actual value of few of the objects like 1F81 & 1F8D of the Controlled Nodes and now the PRes chaining configuration is working fine.
Hence the bug can be closed.

closed 2017-02-03 by zelenkaj