OpenAutomationTechnologies/openPOWERLINK_V2 #187: CMake: static linking with libpcap

RomainNaour, 2016-11-25:
Hello,

When linking demo_mn_console statically with pcap, the CMake build system forget to link with other libraries linked with libpcap (-lnl-genl-3 -lnl-3 -ldbus-1 -pthread).

[100%] Linking C executable demo_mn_console
/home/buildroot/build/instance-0/output/host/usr/x86_64-buildroot-linux-musl/sysroot/usr/lib/../lib64/libpcap.a(pcap-linux.o): In function `nl80211_init':
pcap-linux.c:(.text+0x41e): undefined reference to `nl_socket_alloc'

To fix this, the build system could use pcap-config:
pcap-config --libs --static
-L/path/to/sysroot/usr/lib -lpcap -L/path/to/sysroot/usr/lib/.libs -lnl-genl-3 -lnl-3  -L/path/to/sysroot/usr/lib -ldbus-1 -pthread

Full build log:
http://autobuild.buildroot.net/results/385/3859dc0f4de7e3284a96d5841f040f69f71842df/build-end.log

Best regards,
Romain

banickn, 2016-12-07:
Hello Romain,

I tried to reproduce the issue you reported with the following steps.
At first I compiled libpcap-1.8.1 from source and then changed the cmake file so it links statically to the 'libpcap.a'. This worked for me without any problems. Anyhow, when I run pcap-config --libs --static it doesn't show "-lpcap -L/path/to/sysroot/usr/lib/.libs -lnl-genl-3 -lnl-3 -L/path/to/sysroot/usr/lib -ldbus-1 -pthread" that part of what you reported. Any idea what you have done differently compared to my approach?

Best regards,
Norman

RomainNaour, 2016-12-07:
Hello Norman,

Thanks for looking at this issue.

I really recommend you to use Buildroot for this investigation because this kind of issue may be difficult to reproduce on Linux distribution. The pcap library (even the static one) can be compiled with a different set of dependencies than the one used in Buildroot.

For this case the libpcap.a from Buildroot have more dependencies than the one used in your Linux distribution.
I'm using Debian8 and indeed the pcap library is not linked with dbus and netlink Library.
$ pcap-config --libs --static
-L/usr/lib/x86_64-linux-gnu -lpcap 
It seems Debian used --disable-dbus and --whithout-libnl while configuring libpcap.

So here is the step to reproduce:

1) Clone the Buildroot git repository
$ git clone https://git.buildroot.net/buildroot

2) Checkout the Buildroot git hash used by the autobuilder
see http://autobuild.buildroot.net/results/385/3859dc0f4de7e3284a96d5841f040f69f71842df/gitid
$ git checheckout b14b02698ecb6deb5b2676a0a7747b67ae19d709
Note: You should be able to reproduce the same issue with the current master branch.

3) Download the Buildroot configuration used by the autobuilder
http://autobuild.buildroot.net/results/385/3859dc0f4de7e3284a96d5841f040f69f71842df/config
rename config to .config in the buildroot directory

4) Build the Linux system
make openpowerlink

I hope this help.

Thanks
Romain

banickn, 2016-12-12:
Hi Romain,

I see! To be honest, I am not terribly eager to dive into the whole buildroot workflow.
Nevertheless, we'd appreciate it if you create a pull request with a fix that solves this issue.

Maybe something like piping the output of pcap-config --libs --static into cmake or perhaps you have a more elegant solution already in mind.

Best regards,
Norman

RomainNaour, 2017-05-27:
Hi Norman,

I finally had some time to do it, see
https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/pull/240

Best regards,
Romain

commit dd5d2951 (2017-06-08): [FIX] Use pcap-config to fix static linking

When linking the demo applications statically with pcap,
the CMake build system does not link with other libraries
that are already linked with libpcap
(-lnl-genl-3 -lnl-3 -ldbus-1 -pthread).

To fix this, the build system uses pcap-config:
pcap-config --libs --static

Resolves:
#187

Change-Id: I90d1c8997ca089d5c12dd6bfd276fe04cba572f7
commit 93b25724 (2017-11-24): [FIX] Use pcap-config to fix static linking

When linking the demo applications statically with pcap,
the CMake build system does not link with other libraries
that are already linked with libpcap
(-lnl-genl-3 -lnl-3 -ldbus-1 -pthread).

To fix this, the build system uses pcap-config:
pcap-config --libs --static

Resolves:
#187

Change-Id: I90d1c8997ca089d5c12dd6bfd276fe04cba572f7
closed 2017-06-09 by zelenkaj