OpenAutomationTechnologies/openPOWERLINK_V2 #189: Basic run test fails for Altera Terasic De2-115 (cn-single-hostif-gpio, cn-single-hostif-drv)

RitishaRamesh, 2016-12-08:
### Bug description:
Stack version  : 2.5.0
SHA            : ed45710542594400051c679fcf05f09e44b672bb
Test case      : Basic run test 
Managing Node  : Linux edrv
Controlled Node: Terasic de2-115 (cn-single-hostif-gpio, cn-single-hostif-drv)
Cycle time     : 50ms

### Bug summary:
CN does not initialize and fails to reach operational state.

### Test to reproduce:
1. Build the hardware, stack, driver and application.
2. Run the application with the above mentioned configuration.

zelenkaj, 2016-12-12:
Error log? Is it working with e.g. 1 ms?

powerlink-team-kalycito, 2017-01-30:
No, its not working in 1ms as well.

_**The following log is displayed in the nios terminal when downloading the application elf:**_ 

$ nios2-terminal.exe --instance=0
nios2-terminal: connected to hardware target using JTAG UART on cable
nios2-terminal: "USB-Blaster [USB-0]", device 1, instance 0
nios2-terminal: (Use the IDE stop button or Ctrl-C to terminate)

$----------------------------------------------------
openPOWERLINK embedded CN DEMO application
Using openPOWERLINK stack: V2.5.0
$----------------------------------------------------
NODEID=0x01
Initializing openPOWERLINK stack...

zelenkaj, 2017-01-30:
Thank you for clarification!

powerlink-team-kalycito, 2017-03-25:
This issue could be considered invalid as the issue is seen in only one Altera board. When we tried reproducing this issue, we were not able to reproduce. On analysis, it is found to be a board specific problem and not related to the stack. 

We have verified the basic run test with the openPOWERLINK v2.5.0 stack and hence this bug can be closed.

closed 2017-03-27 by zelenkaj