OpenAutomationTechnologies/openPOWERLINK_V2 #192: SDO over Udp test fails for Linux and Windows design

RitishaRamesh, 2016-12-08:
### Bug description:
Stack version  : 2.5.0
SHA            : ed45710542594400051c679fcf05f09e44b672bb
Test case      : SDO over UDP test
Managing Node  : Linux edrv/Windows
Controlled Node: Bus Controllers
Cycle time     : 10ms

### Bug summary:
SDO over Udp test fails for Linux/Windows designs.

### Steps to reproduce:
1. Build the stack, driver and application.
2. Change the application to perform SDO over Udp test.
3. Rebuild and run the application

Observation:
Udp frames are not received while running the application

zelenkaj, 2016-12-12:
> Udp frames are not received while running the application

Please be more specific!

powerlink-team-kalycito, 2017-03-08:
This issue can be closed. Found that the test case is working fine.

zelenkaj, 2017-03-08:
- I agree for Linux
- How about Windows? SDO/UDP support is missing there...

powerlink-team-kalycito, 2017-06-20:
Verified the issue and it exist in Windows (Terasic De2i-150) design.

powerlink-team-kalycito, 2017-07-03:
On further debugging the issue, we found that SDO_UDP support implementation is available, but disabled by default.

We enabled it by  
- Adding "define CONFIG_INCLUDE_SDO_UDP" in stack/proj/windows/liboplkmnapp-pcieintf/oplkcfg.h

However during stack compilation, we received 103 errors due to missing header file.
- So added '#include "winsock2.h"' in stack/src/user/sdo/sdoudp-windows.c

During application compilation, we receive following linking errors:

> 1>------ Build started: Project: demo_mn_console, Configuration: Release x64 ------
1>     Creating library (openPOWERLINK-dir)/apps/demo_mn_console/build/windows/Release/demo_mn_console.lib and object (openPOWERLINK-dir)/apps/demo_mn_console/build/windows/Release/demo_mn_console.exp
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_bind referenced in function sdoudp_createSocket
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_htonl referenced in function sdoudp_createSocket
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_htons referenced in function sdoudp_createSocket
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_recvfrom referenced in function receiveFromSocket
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_select referenced in function sdoUdpThread
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_sendto referenced in function sdoudp_sendToSocket
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_socket referenced in function sdoudp_createSocket
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_WSAStartup referenced in function sdoudp_initSocket
1>oplkmnapp-pcieintf.lib(sdoudp-windows.obj) : error LNK2019: unresolved external symbol __imp_WSACleanup referenced in function sdoudp_closeSocket
1>(openPOWERLINK-dir)\apps\demo_mn_console\build\windows\Release\demo_mn_console.exe : fatal error LNK1120: 9 unresolved externals
========== Build: 0 succeeded, 1 failed, 2 up-to-date, 0 skipped ==========

We will look into this issue and share more observations.

Melvin-Francis, 2017-09-11:
Fix provided in Pull Request #291 .

seissw, 2017-10-11:
Fixed in 2.7.0-rc1

closed 2017-10-11 by seissw