OpenAutomationTechnologies/openPOWERLINK_V2 #193: Max PDO test fails for Linux 8111 edrv

RitishaRamesh, 2016-12-08:
### Bug description:
Stack version  : 2.5.0
SHA            : ed45710542594400051c679fcf05f09e44b672bb
Test case      : Max PDO test
Managing Node  : Linux 8111 edrv
Controlled Node: Altera Terasic de2-115 Max PDO CN
Cycle time     : 10ms

### Bug summary:
CN doesn't reach operational state and fails to receive Max payload.

### Test to reproduce:
1. Build the stack, driver and application.
2. Change the application to perform Max PDO test.
3. Rebuild and run the application.

### Error log:
EVENT    CFM_PROGRESS   Node=  1, Object 0x1600/001,  123/ 162 Bytes -> SDO Abort=SDO_AC_OBJECT_NOT_MAPPABLE(0x06040041), Error=0x0000
EVENT    CFM_RESULT     Node=  1, NmtNodeCommandConfErr
EVENT    NODE           Node=  1, NmtNodeEventError State:NmtCsNotActive
EVENT    NODE           Node=  1, NmtNodeEventNmtState State:NmtCsNotActive
Node 1 entered state NmtCsNotActive
EVENT    NODE           Node=  1, NmtNodeEventNmtState State:NmtCsPreOperational2
Node 1 entered state NmtCsPreOperational2
EVENT    NODE           Node=  1, NmtNodeEventFound State:NmtCsPreOperational2
Stack found node 1
 VENT    NODE           Node=  1, NmtNodeEventCheckConf State:NmtCsPreOperational2
EVENT    CFM_PROGRESS   Node=  1, Object 0x1011/001,    8/ 162 Bytes -> SDO Abort=SDO_AC_OBJECT_NOT_EXIST(0x06020000), Error=0x0000

zelenkaj, 2016-12-12:
Looks like a wrong configuration, check the SDO aborts!

powerlink-team-kalycito, 2017-03-24:
Yes, the configuration was different. This issue can be closed.

closed 2017-03-29 by seissw