OpenAutomationTechnologies/openPOWERLINK_V2 #195: Compilation test (Driver compilation) fails for Windows NDIS design

RitishaRamesh, 2016-12-08:
### Bug description:
Stack version  : 2.5.0
SHA            : ed45710542594400051c679fcf05f09e44b672bb
Test case      : Compile test (Driver compilation)
Managing Node  : Windows (demo_mn_console)

### Bug summary:
Driver compilation fails for Windows NDIS intermediate driver.

### Steps to reproduce:
1. Compile the driver (drv_ndis_intermediate).

### Error log:
<openPOWERLINK-dir>\drivers\windows\drv_
s_intermediate\drv_ndis_intermediate.vcxproj.met
<openPOWERLINK-dir>\drivers\windows\drv_
s_intermediate\drv_ndis_intermediate.vcxproj" (d
ClCompile target) ->
 <openPOWERLINK-dir>drivers/windows/drv
8): error C2036: 'const void *' : unknown size [
vers\windows\drv_ndis_intermediate\build\drv_ndi
diate.vcxproj]
 <openPOWERLINK-dir>/drivers/windows/drv
 error C4013: 'ctrlk_init' undefined; assuming e
<openPOWERLINK-dir>\drivers\windows\drv_ndis_inter
iate\drv_ndis_intermediate.vcxproj]
 <openPOWERLINK-dir>/drivers/windows/drv
 error C4013: 'ctrlk_exit' undefined; assuming e
<openPOWERLINK-dir>\drivers\windows\drv_ndis_inter
iate\drv_ndis_intermediate.vcxproj]
 <openPOWERLINK-dir>/drivers/windows/drv
 error C4013: 'ctrlk_updateHeartbeat' undefined;
[<openPOWERLINK-dir>\drivers\windows\drv
is_intermediate\drv_ndis_intermediate.vcxproj]

zelenkaj, 2016-12-12:
please format the error log!

zelenkaj, 2016-12-12:
More details about how to reproduce this error needed!

powerlink-team-kalycito, 2017-03-08:
**Steps to reproduce:**

    Build the driver in debug or release mode:    

- In Visual studio command prompt, CD to 'openPOWERLINK-dir/drivers/build/windows/drv_ndis_intermediate/build'.
- Execute the command "cmake -G"NMake Makefiles" -DCMAKE_BUILD_TYPE=Debug/Release .."
- Execute the command "msbuild /t:build /platform=x64 /p:Configuration="Win7 Debug/Release"

**Errors generated:**

1. error C2036: 'const void *' : unknown size in
    openPOWERLINK-dir/drivers/windows/drv_ndis_intermediate/drvintf.c(288):
2. error C4013: 'ctrlk_init' undefined; assuming extern returning int in
    openPOWERLINK-dir/drivers/windows/drv_ndis_intermediate/main.c(244)
3. error C4013: 'ctrlk_exit' undefined; assuming extern returning int in
    openPOWERLINK-dir/drivers/windows/drv_ndis_intermediate/main.c(335)
4. error C4013: 'ctrlk_updateHeartbeat' undefined; assuming extern returning int in
    openPOWERLINK-dir/drivers/windows/drv_ndis_intermediate/main.c(775)

Please find attached for the complete error log:
[Windows_driver_build.txt](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/files/826959/Windows_driver_build.txt)

seissw, 2017-08-11:
After re-validation the issue was found to be solved in 2.6.1.

closed 2017-08-11 by seissw