OpenAutomationTechnologies/openPOWERLINK_V2 #197: User mapping test fails for Windows PCIe design

RitishaRamesh, 2016-12-08:
### Bug description:
Stack version  : 2.5.0
SHA            : ed45710542594400051c679fcf05f09e44b672bb
Test case      : User mapping test
Managing Node  : Windows PCIe Terasic DE2i-150
Controlled Node: Bus Controllers
Cycle time     : 10ms

### Bug summary:
CN's reached operational state but the mapped data is not received by the CN's.

### Test to reproduce:
1. Build the hardware, stack, driver and application.
2. Change the application to perform user mapping test.
3. Rebuild and run the application.

zelenkaj, 2016-12-12:
Please check for configuration error!

powerlink-team-kalycito, 2017-03-08:
Yes, the configuration was different. This issue can be closed.

closed 2017-03-08 by zelenkaj