OpenAutomationTechnologies/openPOWERLINK_V2 #247: Performance degradation issue in Linux designs

powerlink-team-kalycito, 2017-06-09:
### Bug description:

Stack version : v2.6.0-rc1
SHA : b3f8283ff
Test case : Performance test
Managing Node : Linux edrv designs
Controlled Node: Bus controllers

[image]

### Bug summary:

Performance degradation is observed in 'Linux edrv' designs against v2.5.0 performance values.

Also observed below error messages in all configurations:
- EVENT HISTORY HistoryEntry: Type=0x3002 Code=0x8243 - MN operational	
- SDO Abort = SDO_AC_DATA_NOT_TRANSF_DUELOCAL_CONTROL(0x08000021), Error=0x0000 -	MN Not operational
- EVENT HISTORY HistoryEntry: Type:0x8232 (0x1E 00 00 00 00 00 00 00) - MN Not operational

### Steps to reproduce: 

- Build the stack, driver and application
- Run the application with above mentioned configurations

### Expected behavior:
[image]

### Observed behavior:
[image]

testrunner123, 2017-08-15:
Can you, please, describe a little how do you come to performance results. Do you adjust Powerlink cycle time down until error comes from stack or how ?

powerlink-team-kalycito, 2017-08-23:
Hello Testrunner123,

Please find below the configurations used for performance test scenarios:

[image]

**Expected behavior:** All the CNs should be stable for a duration of 8 hours. 
(without any errors like CNs reset, Configuration errors, stability issues)

seissw, 2017-12-15:
The latest measurements with V2.7.0-rc1 did not show any performance problems with the Linux designs, so this issue is closed.

closed 2017-12-15 by seissw