OpenAutomationTechnologies/openPOWERLINK_V2 #318: Binary generation path issue - Windows NDIS

powerlink-team-kalycito, 2017-12-11:
**Bug Description:**
Stack version : v2.7.0-rc1
SHA : c733b294ba07a7813120fe6e4aeac2b46a82d18f
Test case : Basic run
Managing Node : Windows NDIS
Controlled Node: Bus controllers

**Bug Summary:**
Driver and Installer binaries are generated at a different folder("/bin/windows").
(**Excepted Behavior** - The driver and installer binaries should be generated in the folder("bin/windows/amd64")

**Steps to Reproduce:**
- Build the stack and driver
- Build the application
- Verify whether the binaries are generated at the path "bin/windows/"

powerlink-team-kalycito, 2017-12-19:
The fix is given in pull request #324

seissw, 2018-05-25:
Integrated into 2.7.0.

closed 2018-05-25 by seissw