OpenAutomationTechnologies/openPOWERLINK_V2 #330: Startup flag can't go back into Basic Ethernet Mode

wescheng, 2018-01-16:
I was trying go into Basic Ethernet Mode to do some file transfers and then go back to Operational later.

1) I used oplk_writeLocalObject to change NMT_StartUp_U32.Bit13 to 1 before 1st oplk_execNmtCommand(kNmtEventSwReset) and it went to Basic Ethernet Mode
2) Then used oplk_writeLocalObject to change NMT_StartUp_U32.Bit13 to 0 with oplk_execNmtCommand(kNmtEventSwReset) and it went to Operational.
3) Then used oplk_writeLocalObject to change NMT_StartUp_U32.Bit13 to 1 with oplk_execNmtCommand(kNmtEventSwReset) and it still went to Operational.

powerlink-team-kalycito, 2018-03-12:
Hi Wescheng,

Kindly use[ openPOWERLINK forum](https://sourceforge.net/p/openpowerlink/discussion/) to post your query if you are still facing the issue with more detailed information.

Thanks,
Powerlink-Team-Kalycito

wescheng, 2018-03-20:
Hi Powerlink-Team-Kalycito,

Please see following forum post and Wolfgang's suggestion to post here:
https://sourceforge.net/p/openpowerlink/discussion/newbie/thread/d6e8c74d/

Also some further information that I noticed is that even though I can get it start in Basic Ethernet Mode, a kNmtEventSwReset will make it go from Not Active to Pre Operational even when not touching the startup flag.

Thanks,
Wesley

powerlink-team-kalycito, 2018-09-19:
Hi Wescheng,

We have verified the sequence you mentioned i.e., from NMT_MS_BASIC_ETHERNET --> NMT_MS_OPERATIONAL --> NMT_MS_BASIC_ETHERNET and observed that it works fine with oplk_execNmtCommand(kNmtEventSwReset) in the stack.

The bug reported was not reproducible at our end with the given steps (i.e. Using oplk_writeLocalObject and oplk_execNmtCommand(kNmtEventSwReset)) It would be really helpful if you provide us the patch that contains this issue. 

Also provide us following details:
- openPOWERLINK stack version
- MN
- CN

 if you are still facing this issue.

Regards,
Powerlink-Team-Kalycito

powerlink-team-kalycito, 2018-11-28:
The bug is not reproducible in the way user has reported. Hence this bug can be closed.

Regards,
Powerlink-Team-Kalycito