OpenAutomationTechnologies/openPOWERLINK_V2 #335: how can i solving the problem(ZC702 REV1.1 + AES-FMC-ISMNET2-G)?

liuzq71, 2018-03-09:
hello,
going through the steps of the implementation of MN on ZYNQ Hybrid design using openPOWERLINK v2.6.2 http://openpowerlink.sourceforge.net/doc/2.6/2.6.2/df/d2c/page_zynq_hybrid.html, I got these errors, how can i solving it?thanks
[image]

powerlink-team-kalycito, 2018-03-12:
Hello User,

Kindly use [openPOWERLINK forum](https://sourceforge.net/p/openpowerlink/discussion/) to post your query if you are still facing the issue.

Thanks,
Powerlink-Team-Kalycito

liuzq71, 2018-03-12:
but i could not join in  openPOWERLINK forum

liuzq71, 2018-03-12:
[image]

closed 2018-03-12 by liuzq71