OpenAutomationTechnologies/openPOWERLINK_V2 #336: Big-endian support for openMAC driver

sebhub, 2018-03-14:
We would like to use the openMAC on a QorIQ T4240 (big-endian PowerPC) via PCIe.

In omethlib_target.h we have this
```
#if XPAR_MICROBLAZE_ENDIANNESS == 0
    // Microblaze is big endian (with PLB)
    #define OMETH_HW_MODE                    1
#else
    // Microblaze is little endian (with AXI)
    #define OMETH_HW_MODE                     0
#endif
```
The `OMETH_HW_MODE` is only used in one place
```
        // get phy IDs (Reg 2 and 3) to 32Bit var phyId
        // note: big endian reads reg 2 first, otherwise reg 3 is read first
#if OMETH_HW_MODE == 0 //little
        omethPhyRead(hEth, 0, 3, &data);
#else //big
        omethPhyRead(hEth, 0, 2, &data);
#endif
        phyId = data;
#if OMETH_HW_MODE == 0 //little
        omethPhyRead(hEth, 0, 2, &data);
#else //big
        omethPhyRead(hEth, 0, 3, &data);
#endif
        phyId |= ((unsigned long)data << 16);
        phyId = phyId >> 4;    // remove revision number
```
This makes no sense to me. Construction of a 32-bit integer from two 16-bit integers has nothing to do with endianess.

For a real big-endian support we have to use inline functions or macros to read/write the openMAC registers. I would like to introduce new inline functions in omethlib_target.h and use them throughout to access the openMAC registers in the driver. Example for Microblaze target section:
```
static inline UINT8 ometh_rd_8(volatile UINT8 *p)
{
    return Xil_In8((uint32_t)p);
}

static inline void ometh_wr_8(volatile UINT8 *p, UINT8 v)
{
    Xil_Out8((uint32_t)p, v);
}

static inline UINT16 ometh_rd_16(volatile UINT16 *p)
{
    return Xil_In16((uint32_t)p);
}

static inline void ometh_wr_16(volatile UINT16 *p, UINT16 v)
{
    Xil_Out16((uint32_t)p, v);
}

static inline UINT32 ometh_rd_32(volatile UINT32 *p)
{
    return Xil_In32((uint32_t)p);
}

static inline void ometh_wr_32(volatile UINT32 *p, UINT32 v)
{
    Xil_Out32((uint32_t)p, v);
}
```
Example for PowerPC section:
```
static inline UINT8 ometh_rd_8(volatile UINT8 *p)
{
    return *p;
}

static inline void ometh_wr_8(volatile UINT8 *p, UINT8 v)
{
    *p = v;
}

static inline UINT16 ometh_rd_16(volatile UINT16 *p)
{
    return __builtin_bswap16(*p);
}

static inline void ometh_wr_16(volatile UINT16 *p, UINT16 v)
{
    *p = __builtin_bswap16(v);
}

static inline UINT32 ometh_rd_32(volatile UINT32 *p)
{
    return __builtin_bswap32(*p);
}

static inline void ometh_wr_32(volatile UINT32 *p, UINT32 v)
{
    *p = __builtin_bswap32(v);
}
```
The PowerPC is potentially a 64-bit target, so type long may be not 32-bit. We have to adjust the integer types in the openMAC driver to use UINT8, UINT16 and UINT32.

sebhub, 2018-03-14:
The UINT8, UINT16 and UINT32 types are not available in the openMAC driver in general. Can I use the C99 standard header file <stdint.h> in omethlib.h?

powerlink-team-kalycito, 2018-03-16:
Hi Sebastian,

Thanks for pointing it out. Hope you have understood that endianness defined in the code is for Microblaze and not the host processor.
And your approach of using <stdint.h> header is fine.

If you are planning to provide pull request for this specific issue, we will validate and take up the same.

Regards,
Powerlink-Team-Kalycito

sebhub, 2018-03-19:
We want to use the openMAC directly via PCIe from a big-endian 64-bit PowerPC (QorIQ T4240).  The Nios 2 is little-endian. After a review of the OPLK stack we were not sure if a PCP configuration would work with a big-endian host processor and a little-endian PCP.

sebhub, 2018-04-11:
I just notices that I don't follow the commit subject guidelines. Should I recreate the pull request with subjects prefixed by [TASK]?

powerlink-team-kalycito, 2018-04-11:
Hello Sebastian,

Yes, our CI system will throw errors if the coding guidelines is not followed.

Kindly follow the coding guidelines from, http://openpowerlink.sourceforge.net/doc/2.5/2.5.2/page_coding.html 

Here are some commit message guidelines to be followed,

- Header should be tagged with

1. [FEATURE] for new feature addition e.g. new platform or module
2. [FIX] for bug fixes
3. [TASK] for everything else i.e. general task such as adding CMake or updating header and source code after adding a new platform in the previous commit

-  Maximum length of header should be 72 characters including the tag

-  Header message should be in 'simple present' tense
 
-  Commit body can be paragraphs or bulletined.

1. In case of bullets use - followed by a blank space to start the message line

sebhub, 2018-10-08:
All patches for the big-endian support of the openMAC are integrated.

closed 2018-10-08 by sebhub