OpenAutomationTechnologies/openPOWERLINK_V2 #340: Zynq Hybrid - Cann't let Microblaze running issue

liuzq71, 2018-03-30:
openPOWERLINK v2.6.2,vivado 2016.2:

how to let Microblaze run the program with the hardware diagram below?
[image]
i tried it many times but failed every times with the these sentenses:

 Xil_Out32(0xE000A284, 0x1);  //reset Microblaze
 Xil_Out32(0xE000A048, 0x1);

thanks

liuzq71, 2018-04-01:
does the lscript.ld of fsbl.elf must be modified linkly below?
[image]

szaipcljl, 2018-04-02:
Connect the Avnet expander board to the J3 FMC1 connector of the Zynq ZC702 board?

liuzq71, 2018-04-02:
yes,i do ,
now the problem is that i find the Microblze does not run program correctly.

liuzq71, 2018-04-02:
TO szaipcljl:
sir ,would like please give me your email?

liuzq71, 2018-04-04:
Can someone help me?i feel anxious

closed 2018-04-07 by liuzq71

szaipcljl, 2018-04-08:
What is your actual problem?

Closed #340.

liuzq71, 2018-04-08:
是的，按照文档去编译移植，却运行不起来,我的EMAIL:liuzq71@foxmail.com

szaipcljl, 2018-04-08:
你能不能把运行时的情况截图给我看看呀，我看看是不是我之前遇到的问题。因为我也是第一次搞这个powerlink，很多我也不太懂，但是遇到的问题我都解决掉了。发自我的iPhone------------------ Original ------------------From: bruce  <notifications@github.com>Date: Sun,Apr 8,2018 7:29 PMTo: OpenAutomationTechnologies/openPOWERLINK_V2 <openPOWERLINK_V2@noreply.github.com>Cc: szaipcljl <szaipcljl@163.com>, Comment <comment@noreply.github.com>Subject: Re: [OpenAutomationTechnologies/openPOWERLINK_V2] Zynq Hybrid -Cann't let Microblaze run issue (#340)是的，按照文档去编译移植，却运行不起来,我的EMAIL:liuzq71@foxmail.com