OpenAutomationTechnologies/openPOWERLINK_V2 #342: ifconfig is deprecated

RomainNaour, 2018-04-04:
From stack/src/arch/linux/target-linux.c:
The target_setIpAdrs() function still use "ifconfig" deprecated tool to initialize the VETH named plk.
Some new Linux distribution (Debian) doesn't provide "ifconfig" anymore.

A quick fix can be add some code to use ip tool instead.

But, what about using some C code to set the IP address ?

Romain

liuzq71, 2018-04-05:
Did  you build the powerlink follow the document: 
 http://openpowerlink.sourceforge.net/doc/2.6/2.6.2/df/d2c/page_zynq_hybrid.html ?

RomainNaour, 2018-04-05:
No, I don't use Zynq Hybrid design.

The question is not hardware specific.
Under a MN/CN Linux system, OpenPowerLink is using ifconfig tool to configure the Virtual Ethernet Driver (CONFIG_INCLUDE_VETH).

So what happen if ifconfig is missing ?

powerlink-team-kalycito, 2018-09-10:
Pull request [#372](https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2/pull/372) resolves this issue. Hence this bug can be closed.

closed 2018-09-11 by RomainNaour