OpenAutomationTechnologies/openPOWERLINK_V2 #352: Compilation errors for Linux edrv in Ubuntu 18.04

powerlink-team-kalycito, 2018-06-04:
**Bug Description:**
Stack version : v2.7.0
SHA : df70e53c688f56b1
Test case : Compile test
Managing Node : Linux edrv designs

**Test environment details:**
**_Linux edrv:_** i210
**_OS:_** Ubuntu 18.04
**_Kernel version:_** 4.16.7-rt1 

**Bug Summary:**
Linux edrv driver compilation results with errors

**Steps to reproduce:**
Compile the Linux edrv driver

**Error log:**
[ 97%] Generating oplki210mn/src/hrestimer-i210.c
[100%] Generating oplki210mn/oplki210mn.ko
openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn/src/main.c: In function ‘powerlinkOpen’:
openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn/src/main.c:313:5: error: implicit declaration of function ‘init_timer’; did you mean ‘init_timers’? [-Werror=implicit-function-declaration]
     init_timer(&heartbeatTimer_g);
     ^~~~~~~~~~
     init_timers
openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn/src/main.c: In function ‘startHeartbeatTimer’:
openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn/src/main.c:933:31: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
     heartbeatTimer_g.function = increaseHeartbeatCb;
                               ^
openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn/src/main.c:934:21: error: ‘struct timer_list’ has no member named ‘data’
     heartbeatTimer_g.data = 0;
                     ^
cc1: some warnings being treated as errors
scripts/Makefile.build:324: recipe for target '/openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn/src/main.o' failed
make[4]: *** [openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn/src/main.o] Error 1
Makefile:1561: recipe for target '_module_/openPOWERLINK-drivers/linux/drv_kernelmod_edrv/build/oplki210mn' failed
make[3]: *** [_module_openPOWERLINK-dir/drivers/linux/drv_kernelmod_edrv/build/oplki210mn] Error 2
CMakeFiles/oplki210mn.dir/build.make:96: recipe for target 'oplki210mn/oplki210mn.ko' failed
make[2]: *** [oplki210mn/oplki210mn.ko] Error 2
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/oplki210mn.dir/all' failed
make[1]: *** [CMakeFiles/oplki210mn.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

RomainNaour, 2018-07-05:
Hi,

init_timer was removed and replaced by timer_setup since kernel 4.15.
See commit 7eeb6b893bd28c68b6d664de1d3120e49b855cdb

It's not related to Ubuntu 18.04 but all kernel version since 4.15.

Best regards,
Romain

powerlink-team-kalycito, 2018-07-12:
Hi Romain,

Thanks for pointing it out.
We have already fixed the compilation error. We are now working on performance issues seen in the latest kernel.

Thanks,
Powerlink-Team-Kalycito

PR #373 (2018-09-06): [FIX] Fix compilation errors for Linux kernel > v4.15
 - init_timer() is deprecated in new kernels hence add
   timer_setup() to support for kernel > v4.15
 - This fix resolves compilation error for Linux Edrv designs

This pull request supersedes #368. This fix resolves #352
closed 2018-09-06 by zelenkaj